amanusk/s-tui #123: Typo in FAQ

LinuxOnTheDesktop, 2019-07-09:
The FAQ page (here on github) has

> decrees

where

> decrease

is meant.

amanusk, 2019-07-09:
Thanks!
Could you open a PR to make the fix?

LinuxOnTheDesktop, 2019-07-09:
I could if you tell me how to! I'm new to pull requests.

amanusk, 2019-07-09:
Sure:)
This should also be of help: [contributing](https://github.com/amanusk/s-tui/blob/master/CONTRIBUTING.md)

Start by forking the repository you your own account with the `fork` button on the top right. This will create a copy of the repository on your account.
Any changes you make to this repository will only be reflected on your fork an this point.

Next, you can clone the repository to your local device with `git clone <URL of your copy>`.
Now make the relevant changes on your local device.
Next, commit and push the changes to your fork on github.
* In fact, for a simple fix like a readme file, you can edit the files directly online, directly in github, and skip cloning the repository and pushing back the changes.

At this stage, github will probably automatically suggest that you open a pull request when you go to github page of the repository.

If not, follow these steps https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

It might look a bit complected at first, but once you go through the process for the first time, it makes sense :)

LinuxOnTheDesktop, 2019-07-09:
Good explanation (though I'd already cloned and downloaded the repository before I read that I did not have to!).

Done, i.e. pull request generated.

amanusk, 2019-07-11:
I can see that you have created your own fork, but not have opened a pull request yet.

LinuxOnTheDesktop, 2019-07-11:
I thought I followed the instructions. I have tried again. If I have failed, please just fix the typo yourself.

closed 2019-07-11 by LinuxOnTheDesktop