amanusk/s-tui #125: Tempurature graph maxium adjusts to max monitored temp, not cpu high/critical temp

BlisteringSyko, 2019-08-02:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): _____Ubuntu
  * OS version: _____18.04.4
  * s-tui version: _____0.8.3
  * Installation method(pip/PPA/source code): _____sudo pip install s-tui
  

The temp graph seems to change is maximum as the cpu temp increases, this gives the appearance of a constantly maxed out graph, making it not very useful. 

Is there a way to simply set the max value for the graph to 100? 
This would make it visually more useful as you could just glance at the graph and see that it the bars are only 50% the way up, you can estimate the temp. As is i still have to look at the temp in the side bar, defeating the purpose of the graph.

amanusk, 2019-08-07:
@BlisteringSyko, thanks for the issue.
In some previous versions it worked as you suggest. There is a trade-off in "resolution" when going this route. When setting the bar max value high from the start, smaller changes will not be visible.

Also, not all sensors have a `high` and `critical` setting, so we decided to use the highest temperature reached as `max` and raise the bar if it is surpassed. 

There are good arguments both ways, so I am still considering what should be the default behaviour

BlisteringSyko, 2019-08-07:
How about an checkbox or an optional config file/setting?
Maybe a config setting, if a value exists, to use it as the max, if blank use auto?

I didn't really consider the loss of resolution from using a fixed value so now i understand why it is the way it is currently. I feel like it really depends on the end user though, I personally want to use it to glance at, so if see its maxed out id know that its about to catch fire.

highvoltage, 2019-08-08:
I guess this bug is somewhat related, when cpu scaling happens it should probably re-adjust too: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=912783

amanusk, 2019-09-03:
@BlisteringSyko, @highvoltage, I am currently considering the following behaviour:
* Start with reading a "top" measurement where applicable (e.g 100 for utilization) and set it as the max of the graph
* If for some reason this measurement is exceeded, the graph maximum is updated to reflect that.

Makes sense?

This is currently how `master` and the beta in pypi is configured: https://pypi.org/project/s-tui/1.0.0b2/

closed 2020-07-11 by amanusk