amanusk/s-tui #126: Please enable time range so I can monitor a server over hours and days

tomachinz, 2019-08-14:
**Thanks**
This is a great app, I am running it on Linux as my preferred way to graph the CPU!
Perhaps I will try to also help code this feature request and also get the build working on a mac at some point, for now I use this on modern Linux Ubuntu.

**Be able to graph and see time ranges like hours and days**
I'd run s-tui over a tmux connection via ssh. So I can terminate the window and walk away, but then I come back and re-connect using `tmux attach`. Blam I can instantly see a NOC with bandwidth, your CPU graph, and my app etc. 

But I find the time range too small as I am running a long range genomics app - AminoSee DNA viewer - check it. 

*also**
I reckon the arrows <either side> of the menus made me initially and repeatedly press my cursor keys left right over the course of the first weeks of running it. Finally I realised they not do anything and to press enter. Potentially removing those could assist some future users to consider [enter] key as first choice unlike me. Thanks!

amanusk, 2019-08-16:
> Thanks
This is a great app, I am running it on Linux as my preferred way to graph the CPU!
Perhaps I will try to also help code this feature request and also get the build working on a mac at some point, for now I use this on modern Linux Ubuntu.

I'm glad you find it useful! I would be happy to help with on-boarding if you would like to contribute.
There have been some efforts to make temperature sensors work with MacOS, but they were met with some licensing limitations. We already had a working version ;)

> But I find the time range too small as I am running a long range genomics app - AminoSee DNA viewer - check it. Naturally it takes a lot of time to build up that graph - perhaps make it auto-scale if you leave the window running for 5 days you can then see 5 days graph? That would be kind nice!

How do you imagine this working for a period as long as 5 days? I always had something like stress testing for over/underclocking in mind, where results are usually seen after several minutes.
The best thing I could suggest now is to change the update rate, it will then take a long time for the graph to fill up, but you might lose information in-between measurements.

Logging is also there to help record information over extended periods of time, then you need to plot it from the CSV file.

tomachinz, 2019-08-17:
Easy way would be -i 10 to give a 10 second interval? Thats what the other 3 windows are on (htop, and iftop, speedometer).

A nice to have would be if when the graph hits the left side it starts to get squeezed and remains a history from the start? Maybe a flag like --alltime etc.

See how `speedometer` on the right side is showing a large time range with little spikes? Thats the app pushing files around on the LAN cluster! Most tools are running with 10 second update so that could be half an hour on the right side. The s-tui is around every 2 seconds so maybe only shows a minute of time.

Here is a shot from a recent session!
[image]

Anyhow THANKS yeah when I am free I will try to build a macOS version and let you know how I get on!

amanusk, 2019-08-17:
For now, you can change the refresh rate on the right, it is `2.0` by default but this can be edited. Then `<save settings>` to make this persistent.

tomachinz, 2019-08-22:
How does one do such a thing? Is there a CLI parameter for that? Do I need to use the cursor keys to get there?

amanusk, 2019-08-22:
Yes, just navigate to the `Refresh field`, delete with backspace and put in whatever value you want.

[image]

Than go to `<Save settings>`, press enter and it will make it be the defult

tomachinz, 2019-08-26:
Ah yes I see! I need to go [down][down] and then type in 10. Very good.  Thanks!

closed 2019-08-26 by tomachinz