amanusk/s-tui #127: Missing details and dependencies. Can't install.

Kyliael, 2019-08-18:
Lot of info are missing in installation steps.

When I tried to install using  I  got the error importerror "No Module named Setuptools"
Then I had to figureout that you app needs Setuptools.
Since I don't know which one to install I tired to install the python3 version 
sudo apt-get install python3-setuptools
still not working
Then I installed the old one : sudo apt-get install python-setuptools

Now I'm getting another issue, 
 Running setup.py bdist_wheel for s-tui ... error
 error: invalid command 'bdist_wheel'

=> It seems I have now to install this bdist_wheel dependency....
I will now try to install it, but I think that would be really great to add all this depencies stuff in the installation steps to avoid spending so much time searching which one are missing and how to install them one by one.
Thanks

amanusk, 2019-08-18:
Hi @Kyliael, and thanks for opening the issue.
I haven't encountered the error you are describing. It would help  a lot if you describe which OS distribution your are using, and which version of python.
On new ubuntu 18.10 and newer you can install simply with `sudo apt install s-tui`

Otherwise, installing with `pip` should take care of dependencies.

amanusk, 2020-07-11:
Closing this, please reopen if still relevant

closed 2020-07-11 by amanusk