amanusk/s-tui #129: Wrong CPU frequency readings

jirka-h, 2019-08-27:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Lenovo T580 with Intel i7-8650U CPU [1]
  * OS version: Fedora 29
  * s-tui version: 0.8.3
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:

Wrong CPU frequency readings:

Please compare /proc/cpuinfo, Python's psutil output and "s-tui -t" ouput:

$./cpu_freq.py ; grep -i Mhz /proc/cpuinfo ; s-tui -t
scpufreq(current=2700.0278749999998, min=400.0, max=4200.0)
cpu MHz         : 2700.046
cpu MHz         : 2699.998
cpu MHz         : 2700.007
cpu MHz         : 2700.000
cpu MHz         : 2700.027
cpu MHz         : 2699.999
cpu MHz         : 2700.002
cpu MHz         : 2700.010

Top Freq: 4200 MHz, Cur Freq: 2500.0 MHz, Perf Lost: 0 (N/A) run sudo, Max Temp: 67.0 C, Cur Temp: 67.0 C, Utilization: 62.5 %, Max Power: 17.0 W, Cur Power: 16.5 W, Fan: 3940.0 RPM, 

s-tui -t reports 2500.0 MHz, which is a completely different value from what /proc/cpuinfo and Python's psutil report.

Could you please check why we have this discrepancy? 

Also, how s-tui handles systems with more than one CPU? My laptop has 4 cores. (8 CPUs with hyperthreading enabled). Each CPU can run at a different frequency. 

For the reference, here is the Python's psutil script:
$more ./cpu_freq.py
#!/usr/bin/env python2

import psutil
print(psutil.cpu_freq())

[1] https://ark.intel.com/content/www/us/en/ark/products/124968/intel-core-i7-8650u-processor-8m-cache-up-to-4-20-ghz.html

amanusk, 2019-08-27:
s-tui is actually using psutil to gather (most of) the information, including frequency, so there shouldn't be a big difference
psutil, in turn, reads this information from /proc/cpuinfo (most of the time, unless it is unavailable), so that to shouldn't differ too much.
Difference could be the result of timing, the reported frequency can change slightly between measurements.

> Also, how s-tui handles systems with more than one CPU?

psutil has an option to report average CPU usage. It basically takes the sum of all reported CPU cores and divides by the number of cores. This is what s-tui uses in 0.8.3.

In version 1.0, which is currently in beta, there is a separate graph for each core. You can try it by running from source.

amanusk, 2020-01-23:
Version 1.0 now displays frequency per core. @jirka-h, can you validate this fixes your problem?

closed 2020-06-13 by amanusk

jirka-h, 2020-06-22:
Hi Alex,

thanks for fixing it! I have tested version s-tui 1.0.1 and it shows frequency per core now and values are in sync with `/proc/cpuinfo` and `i7z` utility: https://www.linuxlinks.com/i7z/

One suggestion though - consider adding max frequency to the output. I think it's more significant than the average frequency. 

Please check the `i7z` utility as the reference. It reports:

`  Real Current Frequency 2325.16 MHz [100.52 x 23.13] (Max of below)
`
Thanks!
Jirka

PS: My testing results:

Run as root, log in cpu_freq_log.txt file:
$ i7z -w a

Another terminal 
$~/.local/bin/s-tui -t; grep -i Mhz /proc/cpuinfo ; ~/.local/bin/s-tui -t ; tail -1 ~/cpu_freq_log.txt

Frequency: Avg: 2824.9, Core 0: 2758.8, Core 1: 2873.4, Core 2: 2834.2, Core 3: 2784.2, Core 4: 2754.6, Core 5: 2844.0, Core 6: 2915.6, Core 7: 2834.2, Temp: Acpitz,0: 69.0, Iwlwifi_1,0: 33.0, Composite,Pkg0: 28.9, Sensor1,Pkg0: 28.9, Sensor2,Pkg0: 28.9, Pch_Skylake,0: 52.5, Thinkpad,0: 69.0, Thinkpad,1: 0.0, Thinkpad,2: 65.0, Thinkpad,3: 0.0, Thinkpad,4: 0.0, Thinkpad,5: 0.0, Thinkpad,6: 0.0, Thinkpad,7: 0.0, Thinkpad,8: 0.0, Thinkpad,9: 0.0, Thinkpad,10: 0.0, Thinkpad,11: 0.0, Thinkpad,12: 0.0, Thinkpad,13: 0.0, Thinkpad,14: 0.0, Util: Avg: 34.4, Core 0: 0.0, Core 1: 0.0, Core 2: 100.0, Core 3: 0.0, Core 4: 55.6, Core 5: 0.0, Core 6: 0.0, Core 7: 100.0, Power: package-0: 14.8, core,Pkg0: 12.4, uncore,Pkg0: 0.0, dram,Pkg0: 1.1, psys,Pkg0: 25.4, Fan: thinkpad,0: 3561, 

cpu MHz         : 2648.928
cpu MHz         : 2547.714
cpu MHz         : 2713.171
cpu MHz         : 2606.034
cpu MHz         : 2664.552
cpu MHz         : 2480.660
cpu MHz         : 2574.913
cpu MHz         : 2714.241

Frequency: Avg: 2834.2, Core 0: 2761.1, Core 1: 2833.1, Core 2: 2843.0, Core 3: 2884.9, Core 4: 2724.6, Core 5: 2890.3, Core 6: 2889.2, Core 7: 2847.2, Temp: Acpitz,0: 69.0, Iwlwifi_1,0: 32.0, Composite,Pkg0: 28.9, Sensor1,Pkg0: 28.9, Sensor2,Pkg0: 28.9, Pch_Skylake,0: 52.5, Thinkpad,0: 69.0, Thinkpad,1: 0.0, Thinkpad,2: 65.0, Thinkpad,3: 0.0, Thinkpad,4: 0.0, Thinkpad,5: 0.0, Thinkpad,6: 0.0, Thinkpad,7: 0.0, Thinkpad,8: 0.0, Thinkpad,9: 0.0, Thinkpad,10: 0.0, Thinkpad,11: 0.0, Thinkpad,12: 0.0, Thinkpad,13: 0.0, Thinkpad,14: 0.0, Util: Avg: 34.9, Core 0: 12.5, Core 1: 0.0, Core 2: 100.0, Core 3: 0.0, Core 4: 62.5, Core 5: 0.0, Core 6: 0.0, Core 7: 100.0, Power: package-0: 14.4, core,Pkg0: 12.3, uncore,Pkg0: 0, dram,Pkg0: 1.1, psys,Pkg0: 25.2, Fan: thinkpad,0: 3562, 

i7z -w a
1592787282.152836116    2675.617432     2620.509277     2780.106689     2619.987061

amanusk, 2020-07-16:
Thanks, I'm considering how to  add Max temp per sensor. Maybe in a different submenu