amanusk/s-tui #131: README / webpage: multiple problems with one part of the installation instructions

LinuxOnTheDesktop, 2019-08-30:
The README (i.e., at least, https://github.com/amanusk/s-tui) for your wonderful program contains the following.

> (This usuall creates an executable in ~/.local/bin/ dir. Make sure it is in your PATH)

There are multiple problems with this. (1) 'usuall' should be 'usual'. (2) ' dir' (_sic_)? One of my PCs has three directories in `~/.local/bin`, namely, `pip`, `pip3`, `pip3.5`. Which do you mean? Do you mean the 'appropriate' one? Well, which is that? Your recommended installation command - viz., `pip install s-tui --user` - create the directory `~/.local/bin/s-tui` . .

Also: is `sudo install -H s-tui` an alternative way of installing s-tui, and one that avoids these complications?

amanusk, 2019-09-03:
Thanks for the feedback. I'd like to make the installation process as easy as possible.

Installing with `sudo -H pip install s-tui` is an option, and it is also listed. Not everyone likes installing python packages globally as root, the `--user` option installs the package in the user's home directory, which does not require root permissions.

Here is the documentation of `pip install --help`
```
  --user                      Install to the Python user install directory for your platform.
                              Typically ~/.local/, or %APPDATA%\Python on Windows. (See the
                              Python documentation for site.USER_BASE for full details.)
```
After running the command `pip install --user s-tui`, an executable called `s-tui` will be created in the subdirectory called `bin` in `~/.local/`

Thus, to run `s-tui` after installing it with this installation method, you can run the executable `~/.local/bin/s-tui`.

If you want to run the executable simply as `s-tui` from anywhere, you can add the directory to your `PATH` global environment variable.
For example by adding `export PATH=~/.local/bin:$PATH` to the `.bashrc`

Does this make this more clear?

LinuxOnTheDesktop, 2019-09-03:
> Does this make this more clear? Thank you, yes - but with one exception. Let me explain.

You wrote:

> If you want to run the executable simply as s-tui from anywhere, you can add the directory to your PATH global environment variable.
> For example by adding export PATH=~/.local/bin:$PATH to the .bashrc

Yet, `.bashrc` is in `/home/<user>`. So, can that file really enable anyone to run s-tui from anywhere? No. Rather, one will be to the user in question as against some other user, where 'some other user' includes the `su` user.

amanusk, 2019-09-03:
>  So, can that file really enable anyone to run s-tui from anywhere? No. 

This is correct, only the user installing the program will be able to run it from "anywhere". This is the desired effect in this case. 

To install globally for anyone, installation as `su` is required, just as you explained.

Running s-tui as root is optional but not mandatory, the enabled features are explained in the README.

LinuxOnTheDesktop, 2019-09-03:
Right. So you might want to say in the instructions that there are some functions that s-tui can perform only if installed as root. Thanks.

closed 2019-09-03 by LinuxOnTheDesktop