amanusk/s-tui #133: ' RuntimeWarning: ignoring OSError [. .]'

LinuxOnTheDesktop, 2019-09-03:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: Mint 19.2 Cinnamon
  * s-tui version: 0.8.3
  * Installation method(pip/PPA/source code): pip install --user s-tui
  
### Step 2: Describe the problem:

I see this when I run the program:

```
$ s-tui
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1150: RuntimeWarning: ignoring OSError(6, 'No such device or address')
  warnings.warn("ignoring %r" % err, RuntimeWarning)
```

#### Observed Results:

  * What happened? (Traceback of a crash/Image/description): After the warning above is displayed, s-tui does run without any apparent problems.
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2019-09-03 06:57:05,357 [main()] [INFO ]  Started without root permissions
2019-09-03 06:57:05,357 [__init__()] [DEBUG]  Config file not found
2019-09-03 06:57:05,357 [__init__()] [DEBUG]  No refresh rate configed
2019-09-03 06:57:05,357 [__init__()] [DEBUG]  No user config for utf8
2019-09-03 06:57:05,357 [__init__()] [DEBUG]  No user config for temp sensor
2019-09-03 06:57:05,357 [__init__()] [DEBUG]  No user config for temp threshold
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Package id 0
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Core 0
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Core 1
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Core 2
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Core 3
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  
2019-09-03 06:57:05,374 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  
2019-09-03 06:57:05,375 [__init__()] [DEBUG]  Sensor Label
[... 369 lines cut ...]
2019-09-03 06:58:25,924 [update_displayed_graph_data()] [INFO ]  Reading 2.8863046079448176
2019-09-03 06:58:28,068 [update()] [INFO ]  Utilization recorded 5.7
2019-09-03 06:58:28,078 [get_power_usage()] [INFO ]  current 30992887760.0 last 30982039093.0
2019-09-03 06:58:28,078 [get_power_usage()] [INFO ]  Joule_Used 10.848667 seconds_passed 2.159630537033081
2019-09-03 06:58:28,078 [get_power_usage()] [INFO ]  Power reading elapsed
2019-09-03 06:58:28,079 [update()] [INFO ]  Fan speed recorded0.0
2019-09-03 06:58:28,079 [update_displayed_graph_data()] [INFO ]  Reading 823
2019-09-03 06:58:28,079 [update_displayed_graph_data()] [INFO ]  Reading 5.7
2019-09-03 06:58:28,080 [update_displayed_graph_data()] [INFO ]  Reading 47.0
2019-09-03 06:58:28,080 [update_displayed_graph_data()] [INFO ]  Reading 5.023390257716948
2019-09-03 06:58:28,474 [unhandled_input()] [DEBUG]  Caught q
2019-09-03 06:58:28,474 [unhandled_input()] [DEBUG]  None
2019-09-03 06:58:28,474 [kill_child_processes()] [DEBUG]  Killing stress process
2019-09-03 06:58:28,475 [kill_child_processes()] [DEBUG]  No such process
2019-09-03 06:58:28,475 [kill_child_processes()] [DEBUG]  Could not kill process

``

  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. pip install --user s-tui
2. s-tui (~/.local/bin is in my path already, by dint of my .bashrc)

```

LinuxOnTheDesktop, 2019-09-03:
The message affects `glances` too and seems to have to do with sensors. See `https://github.com/nicolargo/glances/issues/1203`.

amanusk, 2019-09-03:
Looks like it is psutil related, both s-tui and glances are using psutil for temperatures. I'll try to look into it

amanusk, 2020-01-23:
Related to #135, now fixed in version 1.0

closed 2020-01-23 by amanusk