amanusk/s-tui #134: 0 is a bad lower value for the temperature graph

jonascarpay, 2019-09-06:
**Is your feature request related to a problem? Please describe.**
The temperature graph has a lower bound of 0 degrees Celsius

**Describe the solution you'd like**
The lower bound is one of
  - the lowest value since program start
  - the lowest value in the current window
  - 0 Kelvin

**Additional context**
The lower bound of 0 for the temperature graph makes little sense. In practical terms, most users are never going to go near 0 degrees, so about half of the resolution of the graph is wasted. In more 'physical' terms, the freezing point of water is completely arbitrary wrt. your computer's temperature, and technically it could even go below it. Contrast this with utilization, which _is_ necessarily a value between 0 and 100.
I think a more sensible lower value would be to just use the lowest value seen so far. The 0 Kelvin lower bound is a joke of course, but at least less arbitrary then 0 degrees Celsius.

amanusk, 2019-10-06:
First of all, thanks for your interest.

Although I agree that 0 is an arbitrary  choice, I find it quite useful.
Most CPUs operate between ~20 to ~100 degrees (depending on ambient temperature), which is comfortably similar to what you would find your water to be in everyday life. I think for the majority of users, a scale to 0->100 degrees Celsius is rather intuitive. 

Choosing say 20 over 0 might make more sense, but would not be any less arbitrary, and would look weird if you actually use water cooling of some other fancy cooling solution. 

Perhaps down the line this could be added to the configuration file.

jonascarpay, 2019-10-06:
What do you think of having it automatically adjust to display the range of observed values? That seems to me to be the most sensible default, I might look into submitting a PR.

amanusk, 2019-10-07:
I think the best option would be to have both static and dynamic options, especially for temperature.
There was a similar discussion regarding max value here in #125

Any PR is appreciated and I would love to help with any questions you have.

taifuranowar, 2024-12-13:
if its possible, can we also have numbers in x axis ? i look at the graph and i have no idea what is the acutal temp is. it can be 70 or 80. its pretty hard to guess.