amanusk/s-tui #135: No such file or directory for file '/sys/class/hwmon/hwmon0/temp1_input'

BjoernDaase, 2019-10-05:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: Fedora 31
  * s-tui version: 0.8.3
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):

Those get spammed onto the console, disrupting the graphs shown:
```
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1222: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon0/temp1_input'
  RuntimeWarning)
```

  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2019-10-05 12:56:57,166 [main()] [INFO ]  Started without root permissions
2019-10-05 12:56:57,167 [__init__()] [DEBUG]  Config file not found
2019-10-05 12:56:57,167 [__init__()] [DEBUG]  No refresh rate configed
2019-10-05 12:56:57,167 [__init__()] [DEBUG]  No user config for utf8
2019-10-05 12:56:57,167 [__init__()] [DEBUG]  No user config for temp sensor
2019-10-05 12:56:57,167 [__init__()] [DEBUG]  No user config for temp threshold
2019-10-05 12:56:57,169 [__init__()] [DEBUG]  stress-ng is not installed
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Package id 0
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Core 0
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Core 1
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Core 2
2019-10-05 12:56:57,175 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,176 [__init__()] [DEBUG]  Core 3
2019-10-05 12:56:57,176 [__init__()] [DEBUG]  Sensor Label
2019-10-05 12:56:57,176 [__init__()] [DEBUG]  
2019-10-05 12:56:57,176 [__init__()] [INFO ]  num cpus 8
2019-10-05 12:56:57,278 [update()] [INFO ]  Utilization recorded 10.3
2019-10-05 12:56:57,279 [__init__()] [DEBUG]  arg temp  None
2019-10-05 12:56:57,279 [init_update()] [DEBUG]  custom temp is None
2019-10-05 12:56:57,289 [init_update()] [DEBUG]  Temperature sensor is set to coretemp
2019-10-05 12:56:57,300 [set_threshold()] [DEBUG]  Temperature threshold set to 100.0
2019-10-05 12:56:57,306 [__init__()] [DEBUG]  Update is updated to <function TemperatureSource.init_update.<locals>.update at 0x7f3d23bedb90>
2019-10-05 12:56:57,307 [get_power_usage()] [INFO ]  current 48337476353.0 last 48337472874.0
2019-10-05 12:56:57,307 [get_power_usage()] [INFO ]  Joule_Used 0.003479 seconds_passed 0.00022363662719726562
2019-10-05 12:56:57,307 [get_power_usage()] [INFO ]  Power reading elapsed
2019-10-05 12:56:57,307 [get_power_usage()] [INFO ]  Max power updated 16
2019-10-05 12:56:57,309 [update()] [DEBUG]  Fan Speend Not Available
2019-10-05 12:56:57,309 [update()] [INFO ]  Fan speed recorded0.0
2019-10-05 12:56:57,320 [on_unicode_checkbox()] [DEBUG]  unicode State is False
2019-10-05 12:56:57,323 [update()] [INFO ]  Utilization recorded 27.8
2019-10-05 12:56:57,326 [get_power_usage()] [INFO ]  current 48337620090.0 last 48337476353.0
2019-10-05 12:56:57,326 [get_power_usage()] [INFO ]  Joule_Used 0.143737 seconds_passed 0.01954817771911621
2019-10-05 12:56:57,326 [get_power_usage()] [INFO ]  Power reading elapsed
2019-10-05 12:56:57,326 [update_displayed_graph_data()] [INFO ]  Reading 1071
2019-10-05 12:56:57,327 [update_displayed_graph_data()] [INFO ]  Reading 27.8
2019-10-05 12:56:57,327 [update_displayed_graph_data()] [INFO ]  Reading 63.0
2019-10-05 12:56:57,327 [update_displayed_graph_data()] [INFO ]  Reading 7.352961593931042
2019-10-05 12:56:57,329 [update()] [INFO ]  Utilization recorded 0.0
2019-10-05 12:56:57,332 [get_power_usage()] [INFO ]  current 48337678684.0 last 48337620090.0
2019-10-05 12:56:57,332 [get_power_usage()] [INFO ]  Joule_Used 0.058594 seconds_passed 0.0055599212646484375
2019-10-05 12:56:57,332 [get_power_usage()] [INFO ]  Power reading elapsed
[... 43 lines cut ...]
2019-10-05 12:57:07,918 [update_displayed_graph_data()] [INFO ]  Reading 2.159271362596997
2019-10-05 12:57:10,015 [update()] [INFO ]  Utilization recorded 16.4
2019-10-05 12:57:10,026 [get_power_usage()] [INFO ]  current 48367450825.0 last 48361259495.0
2019-10-05 12:57:10,026 [get_power_usage()] [INFO ]  Joule_Used 6.19133 seconds_passed 2.1093060970306396
2019-10-05 12:57:10,027 [get_power_usage()] [INFO ]  Power reading elapsed
2019-10-05 12:57:10,027 [update_displayed_graph_data()] [INFO ]  Reading 799
2019-10-05 12:57:10,028 [update_displayed_graph_data()] [INFO ]  Reading 16.4
2019-10-05 12:57:10,030 [update_displayed_graph_data()] [INFO ]  Reading 48.0
2019-10-05 12:57:10,031 [update_displayed_graph_data()] [INFO ]  Reading 2.9352449171392427
2019-10-05 12:57:12,129 [update()] [INFO ]  Utilization recorded 13.7
2019-10-05 12:57:12,141 [get_power_usage()] [INFO ]  current 48371600411.0 last 48367450825.0
2019-10-05 12:57:12,141 [get_power_usage()] [INFO ]  Joule_Used 4.149586 seconds_passed 2.114750623703003
2019-10-05 12:57:12,142 [get_power_usage()] [INFO ]  Power reading elapsed
2019-10-05 12:57:12,142 [update_displayed_graph_data()] [INFO ]  Reading 801
2019-10-05 12:57:12,143 [update_displayed_graph_data()] [INFO ]  Reading 13.7
2019-10-05 12:57:12,144 [update_displayed_graph_data()] [INFO ]  Reading 48.0
2019-10-05 12:57:12,145 [update_displayed_graph_data()] [INFO ]  Reading 1.9622105573535327
2019-10-05 12:57:12,847 [kill_child_processes()] [DEBUG]  Killing stress process
2019-10-05 12:57:12,848 [kill_child_processes()] [DEBUG]  No such process
2019-10-05 12:57:12,848 [kill_child_processes()] [DEBUG]  Could not kill process
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Simply open `s-tui`

amanusk, 2019-10-06:
Seems to be a pstutil bug, could you please share your version of pstuil?

BjoernDaase, 2019-10-06:
Alright
> python -c "import psutil; print(psutil._\_version__)"

5.6.3

amanusk, 2019-10-06:
Could you please list the content of the directory `/sys/class/hwmon/hwmon0/`, and what you get if you run `sensors` (requries lm-sensors to be installed)

BjoernDaase, 2019-10-06:
```
➜  ~ sensors   
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +33.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +32.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +33.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          +8.64 V  
curr1:        +3.07 A  

ath10k_hwmon-pci-0200
Adapter: PCI adapter
temp1:        +42.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +29.5°C  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +25.0°C  (crit = +107.0°C)
```

```
➜  ~ tree /sys/class/hwmon/hwmon0/
/sys/class/hwmon/hwmon0/
├── device -> ../../AC
├── name
├── power
│   ├── autosuspend_delay_ms
│   ├── control
│   ├── runtime_active_time
│   ├── runtime_status
│   └── runtime_suspended_time
├── subsystem -> ../../../../../../../class/hwmon
├── temp1_label
├── temp2_label
└── uevent

3 directories, 9 files
```

bubbleguuum, 2019-10-06:
Same here on openSUSE Tumbleweed, kernel 5.3.1.
I think this is due to the recent kernel.

```
$python3 -c "import psutil; print(psutil.__version__)"
5.6.0
```

```
s-tui
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon0/temp1_input'
  RuntimeWarning)
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon0/temp2_input'
  RuntimeWarning)
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon3/temp1_input'
  RuntimeWarning)
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon3/temp2_input'
  RuntimeWarning)
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring OSError(61, 'No data available') for file '/sys/class/hwmon/hwmon5/temp1_input'
  RuntimeWarning)
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon7/temp1_input'
  RuntimeWarning)
/usr/lib64/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon7/temp2_input'
  RuntimeWarning)
```

```
sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +52.0°C  (high = +100.0°C, crit = +100.0°C)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM
temp1:        +54.0°C  
temp2:        +56.0°C  
temp3:         +0.0°C  
temp4:         +0.0°C  
temp5:         +0.0°C  
temp6:         +0.0°C  
temp7:         +0.0°C  
temp8:         +0.0°C  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +54.0°C  (crit = +128.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:            N/A  

BAT0-acpi-0
Adapter: ACPI interface
in0:         +12.13 V  

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +66.0°C
```

amanusk, 2019-10-06:
@bubbleguuum, so these error did not appear before the update?
If so I might need to install the new kernel on a test system to check this.

There are quite a lot of issues with hwmon in Linux. As an example: https://github.com/giampaolo/psutil/issues/1323

These warnings are displayed each time `s-tui` runs an update cycle?

BjoernDaase, 2019-10-06:
> @bubbleguuum, so these error did not appear before the update?
> If so I might need to install the new kernel on a test system to check this.

Oh, indeed.  
When running a 5.3.x kernel the error messages appear, when running a  5.2.x kernel they don't.

> These warnings are displayed each time `s-tui` runs an update cycle?

I see a lot of them on startup and no pattern when the occur again. But it is way less then every update cycle.

bubbleguuum, 2019-10-06:
I'm confirming that these errors do not happen on Kernel 5.2 (tried 5.2.14)

jiblime, 2019-11-05:
Also the same here.

``` 
$ uname -r
5.3.8
$ python -c "import psutil; print(psutil.__version__)"
5.6.1
```
```
/usr/lib/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon3/temp1_input'
  RuntimeWarning)
/usr/lib/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon3/temp2_input'
  RuntimeWarning)
/usr/lib/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon6/temp1_input'
  RuntimeWarning)
/usr/lib/python3.7/site-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon6/temp2_input'
  RuntimeWarning)
```

```
# cd /sys/class/hwmon/hwmon3
# tree
.
|____uevent
|____power
| |____runtime_active_time
| |____runtime_status
| |____autosuspend_delay_ms
| |____runtime_suspended_time
| |____control
|____temp1_label
|____device
|____temp2_label
|____subsystem
|____name
```

And while running as user or root, `[Errno 5] Input/output error Unable to read file /dev/cpu/0/msr` so max freq. is incorrect

liberodark, 2019-11-29:
same error for me on Manjaro 18.1.3 with 0.8.3 version of s-tui

amanusk, 2019-12-06:
Since this is probably reproducible on certain kernel version, and requires installation of hardware (no temperature sensors in VM) it will take me some time to address this. Any help is appreciated.

amanusk, 2019-12-06:
Ok, dusted out an X220 with Majaro (thanks @liberodark), install kernel 5.3 and was able to reproduce the bug.

It appears that starting 5.3 kernel version, there is a file for (for example) `/sys/class/hwmon/hwmon1/temp1_label` but no corresponding `/sys/class/hwmon/hwmon1/temp1_input`.

Interestingly, there is sometimes `in0_input` which apparently reports voltage, and could be interesting to look at in the future. 

As smartly stated by @giampaolo, Linux's `/sys/class/hwmon` is quite a mess 
> https://github.com/giampaolo/psutil/blob/0e1c78671b0ea0328bc87ea77081ef52e30b1d06/psutil/_pslinux.py#L1218
giampaolo
The immediate action is to ignore the warning emitted by `psutil`.

@bubbleguuum @jiblime @liberodark @BjoernDaase, please confirm fix in latest master

bubbleguuum, 2019-12-06:
Confirming issue is fixed in master

amanusk, 2020-01-23:
Fixed in version 1.0

closed 2020-01-23 by amanusk