amanusk/s-tui #137: Is ubuntu 18.04 still supported?

GM-Script-Writer-62850, 2019-10-14:
I upgraded some hardware opened it and saw it still works, sweet wonder if there are any updates, yep..
I just updated it cleared my old configs and get a error:
```
me@Z97Killer:~/Downloads/s-tui$ python -m s_tui.s_tui
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 927, in <module>
    main()
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 881, in main
    graph_controller = GraphController(args)
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 698, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 631, in _load_config
    FreqSource(),
  File "s_tui/sources/freq_source.py", line 49, in __init__
    self.top_freq = psutil.cpu_freq().max
AttributeError: 'NoneType' object has no attribute 'max'
me@Z97Killer:~/Downloads/s-tui$ python3 -m s_tui.s_tui
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/chad/Downloads/s-tui/s_tui/s_tui.py", line 38, in <module>
    import urwid
ModuleNotFoundError: No module named 'urwid'
```
I need to update my system name...

amanusk, 2019-10-14:
You can install it with `pip3 install urwid` 
Similarly for `psutil`

GM-Script-Writer-62850, 2019-10-14:
got it working in python3, thanks

amanusk, 2019-10-14:
Great. Please update on compatibility issues. Closing

closed 2019-10-14 by amanusk