amanusk/s-tui #144: IndexError: list index out of range

XeonBloomfield, 2020-01-03:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: Ubuntu 18.04.3 LTS
  * s-tui version: 1.0.0b2
  * Installation method(pip/PPA/source code): pip install --pre --user s-tui
  
### Step 2: Describe the problem:
#### Observed Results:

Problem occurs on ThinkPad with Intel and Radeon GPU. When you start s-tui before running any application on dedicated GPU (when it's temperature is not being reported - `sensors` output attached in problem reproduction part) and then start using that card s-tui crashes. Same problem occurs vice versa - if I start s-tui with Radeon GPU active, I can see it's temperature graph and crashes after closing application using dedicated GPU.

  * What happened? (Traceback of a crash/Image/description):

```
/home/xeonbloomfield/.local/lib/python3.6/site-packages/psutil/_pslinux.py:1227: RuntimeWarning: ignoring OSError(22, 'Invalid argument') for file '/sys/class/hwmon/hwmon1/temp1_input'
  RuntimeWarning)
Traceback (most recent call last):
  File "/home/xeonbloomfield/.local/bin/s-tui", line 11, in <module>
    sys.exit(main())
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/s_tui.py", line 882, in main
    graph_controller.main()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/s_tui.py", line 738, in main
    loop.run()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/urwid/main_loop.py", line 823, in _loop
    alarm_callback()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/urwid/main_loop.py", line 173, in cb
    callback(self, user_data)
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/s_tui.py", line 829, in animate_graph
    self.view.update_displayed_information()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/s_tui.py", line 256, in update_displayed_information
    summary.update()
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/sturwid/summary_text_list.py", line 66, in update
    for key, val in self.source.get_summary().items():
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/sources/source.py", line 73, in get_summary
    graph_vector_summary.update(self.get_sensors_summary())
  File "/home/xeonbloomfield/.local/lib/python3.6/site-packages/s_tui/sources/source.py", line 64, in get_sensors_summary
    graph_vector_summary[sub_title_list[graph_idx]] = val_str
IndexError: list index out of range
```
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2020-01-03 13:22:11,425 [_load_config()] [DEBUG]  User refresh rate: 2.0
2020-01-03 13:22:11,425 [_load_config()] [DEBUG]  No user config for temp threshold
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Acpitz,0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name PackageId0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Core0,Pkg0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Core1,Pkg0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Core2,Pkg0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Core3,Pkg0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Iwlwifi,0
2020-01-03 13:22:11,430 [__init__()] [DEBUG]  Temp sensor name Pch_Cannonlake,0
2020-01-03 13:22:11,582 [__init__()] [DEBUG]  Fan sensor name thinkpad,0
2020-01-03 13:22:11,584 [__init__()] [INFO ]  num cpus 8
2020-01-03 13:22:11,616 [on_unicode_checkbox()] [DEBUG]  unicode State is True
2020-01-03 13:22:11,620 [main_window()] [DEBUG]  Pile index: 18
2020-01-03 13:22:11,627 [eval_hooks()] [DEBUG]  Evaluating hooks
2020-01-03 13:22:11,630 [update()] [INFO ]  Core id util 8.3
2020-01-03 13:22:11,630 [update()] [INFO ]  Core id util 30.4
2020-01-03 13:22:11,630 [update()] [INFO ]  Core id util 8.7
2020-01-03 13:22:11,630 [update()] [INFO ]  Core id util 4.3
2020-01-03 13:22:11,630 [update()] [INFO ]  Core id util 4.3
2020-01-03 13:22:11,630 [update()] [INFO ]  Core id util 0.0
[... 125 lines cut ...]
power1:           N/A  (cap =  25.00 W)
```

Sensors output with Radeon GPU enabled:
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +47.0°C  (high = +100.0°C, crit = +100.0°C)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:       +1.07 V  
temp1:        +48.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:        7.05 W  (cap =  25.00 W)
```

If you need any help resolving this issue, let me know.

amanusk, 2020-01-03:
Interesting but. Thanks.
The problem is that at the moment there is poor support for the number of sensors changing during the operation. Basically we assume that the number of sensors on `init` will remain the same, but interestingly, the GPU is being toggled on and off depending on the running application. 
This might take some time to fix, to properly be ready for changing sensors per measurement.

XeonBloomfield, 2020-01-04:
Right now version 1.0.0b (as well as the latest code from master branch) is unusable on that hardware when this situation occurs, however 0.8.3 works great but lacks new features. 

What comes to my mind as a quick fix for this problem:

- in case of sensor appearing -> ignore that sensor
- in case of sensor disappearing -> zero it's value 

This prevents s-tui from crashing and gives time to do handling of changing sensors per measurement later. If you like that solution, I can do pull request with implementation.

amanusk, 2020-01-05:
0.8.3 should be working, as it only has a single sensors measured for each category. So this makes sense.
I think your solution is a good one for a quick fix.
For the longer term, we should move away from using a list for the subgraphs and use dictionaries instead. That way only entries with existing values can be populated, with an optional button to refresh the entire view.

HughWarrington, 2020-06-09:
Related? https://bugs.launchpad.net/ubuntu/+source/s-tui/+bug/1882846

amanusk, 2020-06-10:
Thanks @HughWarrington. Yes this is related. s-tui does not handle a change in the number of sensors during operation gracefully. I'll do my best to look into it promptly as it appears more widespread that I initially thought.

amanusk, 2020-06-13:
@HughWarrington, I have pushed an update that should fix this bug.
There is more work to do, but it should at least prevent the crashes.
Can you please try running from source or installing version `1.0.1` from pip and see if it fixes the issue for you?

sarthakmalik, 2020-06-14:
@amanusk Thanks for the amazing work squashing thread crashes. Just tested ``1.0.1`` from pip and it's been very stable in handling sensors hot swap operations.

HughWarrington, 2020-06-16:
@amanusk many thanks for the quick response. I've tried running 1.0.1 from pip, and no problems so far after running for a day in the same environment as before.

closed 2020-07-11 by amanusk