amanusk/s-tui #151: Temperature threshold is not run

bennaux, 2020-04-03:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel, Notebook
  * OS version: Ubuntu 18.04 Live DVD
  * s-tui version: 1.0.0
  * Installation method(pip/PPA/source code): sudo pip install s-tui
  
### Step 2: Describe the problem:

I've placed a file `~/.config/s-tui/hooks.d/temperaturesource.sh`:

    #!/bin/bash
    echo "$(date +%Y-%m-%d_%H:%M:%S) EMERGENCY KILL BECAUSE OF HIGH TEMPERATURE" >> ~/stresstest.txt
    killall stress
    killall FIRESTARTER
    killall mprime

Made it executable: `chmod +x ~/.config/s-tui/hooks.d/temperaturesource.sh`

And invoked s-tui like this:

    s-tui --csv --no-mouse -tt 40

Started `stress`:

    stress --cpu 4 --timeout 20s

The temperatures shoot right over the given 40 degrees, but the stress command always terminates itself after the 20 seconds and is not killed by the threshold script.

bennaux, 2020-04-03:
I've tried to call it `tempsource.sh` as well, with no effects... 

What am I doing wrong?

amanusk, 2020-07-03:
Hey, 
Thanks for the bug report. This is indeed a somewhat untainted feature. It was relevant when s-tui was only displaying a single temperature sensor. Now that it displays all, I'm not sure how to build the feature so the alert will only fire for a specific sensor, for for several.

bennaux, 2020-07-05:
Maybe it would be possible to split it up somehow?

A minimal approach could be to just compare max(sensor1, sensor2, ...) with the threshold.

amanusk, 2020-07-11:
Indeed, the minimal approach is what we'll use for now.
I've pushed a fix and and update for the readme. The file indeed needs to be called `tempsource.sh`.
Regardless, a fix was needed to correctly respect the passed parameters for the threshold temperature (`-tt`)

amanusk, 2020-07-12:
I pushed a fix in version 1.0.2. Please verify. Thank you

closed 2020-07-26 by amanusk

bennaux, 2020-07-26:
Thank you very much -- I do not have any machine to verify it at the moment, but I will have one in a few months and try it. :-)