amanusk/s-tui #154: Crashes upon launch on a virtual machine [`psutil.cpu_freq()` returns `None`]

akhan3, 2020-06-18:
### Step 1: Describe your environment

  * System: Intel Xeon E5-2676 v3 @ 2.4GHz on an AWS EC2 virtual machine
  * OS version: Ubuntu 20.04 x86_64 Kernel 5.4.0-1015-aws
  * s-tui version: 1.0.0-1
  * Installation method: `apt install s-tui` from Ubuntu 20.04 repo
  
### Step 2: Describe the problem:
The program crashes upon launch.

#### Observed Results:

The problem seems to be that `psutil.cpu_freq()` returns `None` on a VM.

```
$ s-tui
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 11, in <module>
    load_entry_point('s-tui==1.0.0b3', 'console_scripts', 's-tui')()
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 881, in main
    graph_controller = GraphController(args)
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 698, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 631, in _load_config
    FreqSource(),
  File "/usr/lib/python3/dist-packages/s_tui/sources/freq_source.py", line 49, in __init__
    self.top_freq = psutil.cpu_freq().max
AttributeError: 'NoneType' object has no attribute 'max'
```
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2020-06-18 20:14:51,711 [_load_config()] [DEBUG]  Config file not found
2020-06-18 20:14:51,711 [_load_config()] [DEBUG]  No refresh rate configed
2020-06-18 20:14:51,711 [_load_config()] [DEBUG]  No user config for utf8
2020-06-18 20:14:51,711 [_load_config()] [DEBUG]  No user config for temp threshold
2020-06-18 20:15:59,930 [_load_config()] [DEBUG]  Config file not found
2020-06-18 20:15:59,930 [_load_config()] [DEBUG]  No refresh rate configed
2020-06-18 20:15:59,930 [_load_config()] [DEBUG]  No user config for utf8
2020-06-18 20:15:59,930 [_load_config()] [DEBUG]  No user config for temp threshold
2020-06-18 20:16:15,978 [_load_config()] [DEBUG]  Config file not found
2020-06-18 20:16:15,978 [_load_config()] [DEBUG]  No refresh rate configed
2020-06-18 20:16:15,978 [_load_config()] [DEBUG]  No user config for utf8
2020-06-18 20:16:15,979 [_load_config()] [DEBUG]  No user config for temp threshold
```
  
### Step 3: Reproduce the problem:

```
$ ipython3 --pdb $(which s-tui)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/usr/bin/s-tui in <module>
      9     sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
     10     sys.exit(
---> 11         load_entry_point('s-tui==1.0.0b3', 'console_scripts', 's-tui')()
     12     )

/usr/lib/python3/dist-packages/s_tui/s_tui.py in main()
    879 
    880     global graph_controller
--> 881     graph_controller = GraphController(args)
    882     graph_controller.main()
    883 

/usr/lib/python3/dist-packages/s_tui/s_tui.py in __init__(self, args)
    696         self.temp_thresh = None
    697 
--> 698         possible_sources = self._load_config(args.t_thresh)
    699 
    700         # Needed for use in view

/usr/lib/python3/dist-packages/s_tui/s_tui.py in _load_config(self, t_thresh)
    629         # This should be the only place where sources are configured
    630         possible_sources = [TempSource(self.temp_thresh),
--> 631                             FreqSource(),
    632                             UtilSource(),
    633                             RaplPowerSource(),

/usr/lib/python3/dist-packages/s_tui/sources/freq_source.py in __init__(self)
     47             self.last_measurement.append(0)
     48 
---> 49         self.top_freq = psutil.cpu_freq().max
     50         self.max_freq = self.top_freq
     51 

AttributeError: 'NoneType' object has no attribute 'max'
> /usr/lib/python3/dist-packages/s_tui/sources/freq_source.py(49)__init__()
     47             self.last_measurement.append(0)
     48 
---> 49         self.top_freq = psutil.cpu_freq().max
     50         self.max_freq = self.top_freq
     51 

ipdb> print(psutil.cpu_freq())   
None
ipdb>  

```

akhan3, 2020-06-18:
This bug probably belongs to the upstream project https://github.com/giampaolo/psutil. Thoughts?

amanusk, 2020-07-03:
Hey, thanks for the detailed report.
Could you check what `psutil.cpu_freq()` returns when run it without s-tui in a vm.
Running something like this:
`python3 -c "import psutil; print(psutil.cpu_freq())"`

akhan3, 2020-07-03:
On a physical Ubuntu 20.04 system, 
```
$ python3 -c "import psutil; print(psutil.cpu_freq())"
scpufreq(current=2369.3248750000002, min=800.0, max=4000.0)
```

amanusk, 2020-07-03:
Thanks. What does it return on the VM you are trying to run s-tui on?

akhan3, 2020-07-03:
```
aamir@ip-172-31-34-48:~$ sudo apt install -y python3-psutil           
aamir@ip-172-31-34-48:~$ python3 -c "import psutil; print(psutil.cpu_freq())"
None
```

amanusk, 2020-07-03:
Thanks, that should be an easy fix. Can you please check if it works on branch `fix_unsaved_summaries`

akhan3, 2020-07-03:
Same error. 
```
aamir@ip-172-31-34-48 (fix_unsaved_summaries) ~/work/s-tui/build/lib$ python3 -m s_tui.s_tui
WARNING:root:Failed to find or create scripts directory,                             proceeding without scripting support
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 947, in <module>
    main()
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 901, in main
    graph_controller = GraphController(args)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 713, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 646, in _load_config
    FreqSource(),
  File "/home/aamir/work/s-tui/build/lib/s_tui/sources/freq_source.py", line 50, in __init__
    self.top_freq = psutil.cpu_freq().max
AttributeError: 'NoneType' object has no attribute 'max'
```

amanusk, 2020-07-03:
Thanks. Just to make sure, what is last commit you are working with?

Also, what vm manager and are you using and what os are you running?

akhan3, 2020-07-03:
I am working with commit `7260a6e621e199e0bb3e9562de7289f94e824c5f`.
I am using Ubuntu 20.04 on an AWS EC2 virtual machine.

```
$ lscpu  | grep -i hypervisor 
Hypervisor vendor:               Xen
```

Following is the build log.
```
aamir@ip-172-31-34-48 (fix_unsaved_summaries) ~/work/s-tui$ git show -s
commit 7260a6e621e199e0bb3e9562de7289f94e824c5f (HEAD -> fix_unsaved_summaries, origin/fix_unsaved_summaries)
Author: Alex Manuskin <amanusk@tuta.io>
Date:   Fri Jul 3 18:46:51 2020 +0300

    Only check not none if attr exists

aamir@ip-172-31-34-48 (fix_unsaved_summaries) ~/work/s-tui$ git status 
On branch fix_unsaved_summaries
Your branch is up to date with 'origin/fix_unsaved_summaries'.

nothing to commit, working tree clean

aamir@ip-172-31-34-48 (fix_unsaved_summaries) ~/work/s-tui$ python3 setup.py build
running build
running build_py
creating build
creating build/lib
creating build/lib/s_tui
copying s_tui/help_menu.py -> build/lib/s_tui
copying s_tui/stress_menu.py -> build/lib/s_tui
copying s_tui/__init__.py -> build/lib/s_tui
copying s_tui/about_menu.py -> build/lib/s_tui
copying s_tui/s_tui.py -> build/lib/s_tui
copying s_tui/helper_functions.py -> build/lib/s_tui
copying s_tui/sensors_menu.py -> build/lib/s_tui
creating build/lib/s_tui/sources
copying s_tui/sources/temp_source.py -> build/lib/s_tui/sources
copying s_tui/sources/source.py -> build/lib/s_tui/sources
copying s_tui/sources/rapl_power_source.py -> build/lib/s_tui/sources
copying s_tui/sources/util_source.py -> build/lib/s_tui/sources
copying s_tui/sources/rapl_read.py -> build/lib/s_tui/sources
copying s_tui/sources/fan_source.py -> build/lib/s_tui/sources
copying s_tui/sources/__init__.py -> build/lib/s_tui/sources
copying s_tui/sources/hook.py -> build/lib/s_tui/sources
copying s_tui/sources/hook_script.py -> build/lib/s_tui/sources
copying s_tui/sources/script_hook_loader.py -> build/lib/s_tui/sources
copying s_tui/sources/freq_source.py -> build/lib/s_tui/sources
creating build/lib/s_tui/sturwid
copying s_tui/sturwid/complex_bar_graph.py -> build/lib/s_tui/sturwid
copying s_tui/sturwid/ui_elements.py -> build/lib/s_tui/sturwid
copying s_tui/sturwid/__init__.py -> build/lib/s_tui/sturwid
copying s_tui/sturwid/summary_text_list.py -> build/lib/s_tui/sturwid
copying s_tui/sturwid/bar_graph_vector.py -> build/lib/s_tui/sturwid

aamir@ip-172-31-34-48 (fix_unsaved_summaries) ~/work/s-tui$ cd ~/work/s-tui/build/lib/

aamir@ip-172-31-34-48 (fix_unsaved_summaries) ~/work/s-tui/build/lib$ python3 -m s_tui.s_tui
WARNING:root:Failed to find or create scripts directory,                             proceeding without scripting support
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 947, in <module>
    main()
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 901, in main
    graph_controller = GraphController(args)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 713, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 646, in _load_config
    FreqSource(),
  File "/home/aamir/work/s-tui/build/lib/s_tui/sources/freq_source.py", line 50, in __init__
    self.top_freq = psutil.cpu_freq().max
AttributeError: 'NoneType' object has no attribute 'max'

```

amanusk, 2020-07-04:
Thanks, I am not able to reproduce on a VM with 18.04. 
I'll try to see if I can reproduce on 20.04 when I get a chance.

If you find a fix, feel free to open a PR

akhan3, 2020-07-06:
I tried commit 11cf46098bd1c263b3b27d52d876069e5e0bfeec on 18.04 VM and experience the same error.

```
$ python3 -m s_tui.s_tui
WARNING:root:Failed to find or create scripts directory,                             proceeding without scripting support
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 947, in <module>
    main()
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 901, in main
    graph_controller = GraphController(args)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 713, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/home/aamir/work/s-tui/build/lib/s_tui/s_tui.py", line 646, in _load_config
    FreqSource(),
  File "/home/aamir/work/s-tui/build/lib/s_tui/sources/freq_source.py", line 50, in __init__
    self.top_freq = psutil.cpu_freq().max
AttributeError: 'NoneType' object has no attribute 'max'
```

amanusk, 2020-07-11:
Can you please elaborate the step you made to get this
I have just started a new 18.04 vm, and did the following steps:
`sudo apt install python3-pip`
`pip install psutil`
`pip install urwid` 

Clone s-tui, run with `python3 -m s_tui.s_tui`

s-tui starts and works.

nickdnk, 2020-09-30:
I have this problem on 20.04 bare-metal. No idea how to proceed. What can I do to debug?

amanusk, 2020-10-01:
@nickdnk can you share more details, s-tui version, hardware etc?

nickdnk, 2020-10-01:
Yes. This is on an Intel 9600K with an MSI motherboard. I run the Liquorix kernel (https://liquorix.net/) version 5.8. The s-tui version is just the latest available from the ubuntu repos using `apt install s-tui`.

amanusk, 2020-10-01:
Please try to run the latest version available with pip. (`pip install s-tui --user`)
If this does not work, please paste the error message here

nickdnk, 2020-10-01:
That worked (using pip3 though). But it's very, very slow. Is that normal? Navigating the app is almost impossible due to latency.

amanusk, 2020-10-02:
No, this is unusual. If you don't mind please share a screen record.
Also, maybe try another terminal. I haven't encountered that before. Consider opening a separate issue.

grit96, 2020-11-10:
On my physical Ubuntu 20.04 machine `python3 -c "import psutil; print(psutil.cpu_freq())"` returns `None` also.

wnz99, 2021-02-27:
I had a similar issue on Ubuntu 20.04 and s-tui 1.0.0-beta3 running on a barebone machine and it turned out that the error was caused by disabling `AMD Cool'n'Quiet ` in the bios.

amanusk, 2021-02-27:
The `1.0.0-beta3 ` version that comes bundled with ubuntu 20.04 had some bugs. Some of them were solved in later versions. 
Can you please try running 1.0.2 from source and report if you have similar issues?

wnz99, 2021-02-27:
I tried `s-tui 1.1.0` and it works fine. Thank you.

amanusk, 2021-02-27:
That's great. Thanks. Closing for now

closed 2021-02-27 by amanusk

HonestQiao, 2022-06-06:
```
$ cat /etc/issue
Debian GNU/Linux bookworm/sid \n \l

$ uname -a
Linux Debian-StarFive 5.17.0-rc4-starfive-rc4 #1 SMP PREEMPT Wed Feb 16 00:47:02 CST 2022 riscv64 GNU/Linux

$ sudo apt install s-tui stress
$ s-tui
```

> Traceback (most recent call last):
>   File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
>     exec(code, run_globals)
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 958, in <module>
>     main()
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 912, in main
>     graph_controller = GraphController(args)
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 716, in __init__
>     possible_sources = self._load_config(args.t_thresh)
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 649, in _load_config
>     FreqSource(),
>   File "/usr/lib/python3/dist-packages/s_tui/sources/freq_source.py", line 50, in __init__
>     self.top_freq = psutil.cpu_freq().max
> AttributeError: 'NoneType' object has no attribute 'max'

```
$ sudo pip install psutil
$ sudo pip install urwid
$ s-tui
```

> Traceback (most recent call last):
>   File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
>     exec(code, run_globals)
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 958, in <module>
>     main()
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 912, in main
>     graph_controller = GraphController(args)
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 716, in __init__
>     possible_sources = self._load_config(args.t_thresh)
>   File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 649, in _load_config
>     FreqSource(),
>   File "/usr/lib/python3/dist-packages/s_tui/sources/freq_source.py", line 50, in __init__
>     self.top_freq = psutil.cpu_freq().max
> AttributeError: 'NoneType' object has no attribute 'max'

electricalcoder, 2025-04-26:
I met the same error，and I tried the method which is as follows to solve it:

which s-tui  # for example, the output is：/home/user/.local/bin/s-tui

sudo ln -s ~/.local/bin/s-tui /usr/local/bin/s-tui

sudo s-tui