amanusk/s-tui #158: Summaries selection not saved

drjhe, 2020-07-01:
The title says it all really: the selection of Summaries is not saved (or restored) with the Save Settings function.

s-tui 1.0.1 installed from pip.

**To reproduce the issue:**

1. Start s-tui.
2. Modify the Summaries selection.
3. Select Save Settings.
4. Quit s-tui.
5. Start s-tui. Note all Summaries are present.

amanusk, 2020-07-03:
Hey, thanks for the report.
I'll take a look at it.
Meanwhile, you could manually edit the config file. It is automatically generated at `~/.config/s-tui/s-tui.conf`.

After the first time you save settings, there will be a line for each graph and summary which you can change to `False`

amanusk, 2020-07-11:
Fix for unsaved summarise is now in master, please check if this works for you.

drjhe, 2020-07-12:
Confirmed fixed in 1.0.2. Thanks!

closed 2020-07-12 by drjhe