amanusk/s-tui #161: CSV file does not contain frequency

julprov, 2020-07-20:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: Ubuntu 20.04
  * s-tui version: 1.0.0.1
  * Installation method(pip/PPA/source code): apt install
  
### Step 2: Describe the problem:
#### Observed Results:
I am trying to save the data from s-tui  in .csv. I am using this command :  
s-tui --csv-file StressData.csv --csv
I can see the frequency in the interface. However, it is not being saved in the .csv file. Is there something I can change to obtain this ?

Thank you !

amanusk, 2020-07-21:
Thanks, that's probably a bug. I'll look into it

joost14, 2021-11-18:
@amanusk could you apply this fix? I am also getting this problem and I am having difficulties installing the fix (bit new to github and linux)

Thank you!

akostadinov, 2024-04-03:
I see frequency in `s-tui-1.1.6-1.fc39.noarch`, perhaps this can be closed.