amanusk/s-tui #164: AMDGPU Power reporting

schaerfo, 2020-08-02:
I would like to add reporting of GPU power from the AMDGPU driver [1] in the power section of s-tui, would you be interested in that?
However, since the driver reports power directly in Watts (instead of the energy consumed from which the power can be calculated by differentiation), some refactoring would be required.

[1] https://www.kernel.org/doc/html/latest/gpu/amdgpu.html#hwmon-interfaces

amanusk, 2020-09-12:
I've been thinking about adding GPU support in general, but I don't want to create a mix of CPU and GPU stats.

I have implemented the basic functionality for Nvidia GPUs here: https://github.com/amanusk/s-tui/tree/nv-tui 

Would be awesome to have a similar implementation for multiple AMD GPUs. 

My dilemma is whether to have this as a separate project or simply to switch between GPU/CPU via a config. Would love to hear inputs

schaerfo, 2020-09-25:
I think it would be most useful to be able to monitor CPU and GPU simultaneously, perhaps in separate `s-tui` instances. Would that be possible/convenient with a configuration option? I can also imagine to select between CPU and GPU modes at startup with a command line option.

robertobarreiro, 2023-04-10:
AMD gpu fans and temps can be added, it would be great to do the same with power.