amanusk/s-tui #167: No power usage readings / Permission denied

galfwender, 2020-11-12:
When running as a normal user, s-tui does not show power usage and upon exiting the following errors are displayed in the terminal:
```
WARNING:root:ignoring (PermissionError(13, 'Permission denied'), '/sys/class/powercap/intel-rapl:0/') for file <class 'RuntimeWarning'>
WARNING:root:ignoring (PermissionError(13, 'Permission denied'), '/sys/class/powercap/intel-rapl:0:0/') for file <class 'RuntimeWarning'>
WARNING:root:ignoring (PermissionError(13, 'Permission denied'), '/sys/class/powercap/intel-rapl:0:1/') for file <class 'RuntimeWarning'>
WARNING:root:ignoring (PermissionError(13, 'Permission denied'), '/sys/class/powercap/intel-rapl:0:2/') for file <class 'RuntimeWarning'>
WARNING:root:ignoring (PermissionError(13, 'Permission denied'), '/sys/class/powercap/intel-rapl:1/') for file <class 'RuntimeWarning'>
```

Running as root works as expected, showing power usage without error. Arch 5.9.8-arch1-1.

amanusk, 2020-11-13:
Thanks for opening the issue. I wonder if this is related to this https://platypusattack.com

Perhaps on newer kernels the permission to read the rapl files has been locked to root users only.
This is a shame, but the info would still be accessible when running as root as you suggested

ejgallego, 2022-08-24:
Duplicate of #179

closed 2025-05-13 by amanusk