amanusk/s-tui #168: Dectecting only 8/16 cores (sort of)

cullyn-inverba, 2020-12-18:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc):  AMD
  * OS version: Arch Linux
  * s-tui version: s-tui 1.0.2 
  * Installation method(pip/PPA/source code): pacman
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description): Detecting only 8 cores (sort of)
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
2020-12-18 14:07:01,172 [_load_config()] [DEBUG]  User refresh rate: 0.5
2020-12-18 14:07:01,172 [_load_config()] [DEBUG]  No user config for temp threshold
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Edge,0
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Junction,0
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Mem,0
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Tctl,0
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Tdie,0
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Tccd1,0
2020-12-18 14:07:01,191 [__init__()] [DEBUG]  Temp sensor name Composite,0
2020-12-18 14:07:01,458 [__init__()] [DEBUG]  Power reading is not available
2020-12-18 14:07:01,459 [__init__()] [DEBUG]  Fan sensor name amdgpu,0
2020-12-18 14:07:01,460 [__init__()] [INFO ]  num cpus 16
2020-12-18 14:07:01,473 [on_unicode_checkbox()] [DEBUG]  unicode State is True
2020-12-18 14:07:01,477 [main_window()] [DEBUG]  Pile index: 18
2020-12-18 14:07:01,491 [eval_hooks()] [DEBUG]  Evaluating hooks
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 2.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 2.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 22.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 5.6
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 2.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 2.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 2.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 2.9
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 5.7
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 3.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:01,494 [update()] [INFO ]  Utilization recorded [3.1, 2.9, 2.9, 22.9, 5.6, 2.9, 0.0, 2.9, 0.0, 2.9, 2.9, 5.7, 3.0, 0.0, 0.0, 0.0, 0.0]
2020-12-18 14:07:01,495 [update()] [INFO ]  Reading [58.0, 58.0, 58.0, 54.5, 54.5, 43.0, 45.85]
2020-12-18 14:07:01,508 [update()] [INFO ]  Reading [2213.8234375, 2326.188, 2141.995, 1920.741, 1944.258, 1951.832, 2537.585, 2160.73, 2194.079, 2768.772, 2335.198, 1949.026, 1936.218, 2192.567, 2685.189, 2189.909, 2186.888]
2020-12-18 14:07:01,508 [get_top()] [DEBUG]  Returning top 3600.0
2020-12-18 14:07:01,510 [update()] [INFO ]  Reading [3.1, 2.9, 2.9, 22.9, 5.6, 2.9, 0.0, 2.9, 0.0, 2.9, 2.9, 5.7, 3.0, 0.0, 0.0, 0.0, 0.0]
2020-12-18 14:07:02,025 [eval_hooks()] [DEBUG]  Evaluating hooks
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 10.9
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 5.5
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 3.6
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 3.6
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 5.7
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 3.7
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 3.8
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 3.7
2020-12-18 14:07:02,027 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:02,028 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:02,028 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:02,028 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:02,028 [update()] [INFO ]  Utilization recorded [2.9, 1.9, 1.9, 10.9, 5.5, 1.9, 3.6, 1.9, 3.6, 5.7, 3.7, 3.8, 3.7, 1.9, 0.0, 0.0, 0.0]
2020-12-18 14:07:02,028 [update()] [INFO ]  Reading [58.0, 58.0, 58.0, 53.625, 53.625, 42.75, 45.85]
2020-12-18 14:07:02,028 [update()] [INFO ]  Reading [2197.9478750000003, 2719.648, 2267.776, 1997.357, 1952.12, 1989.274, 2619.528, 2005.703, 2123.775, 2993.726, 2253.286, 1918.79, 1973.683, 1927.562, 2419.827, 1956.624, 2048.487]
[... 91 lines cut ...]
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 3.8
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 5.6
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 1.9
2020-12-18 14:07:04,611 [update()] [INFO ]  Core id util 0.0
2020-12-18 14:07:04,611 [update()] [INFO ]  Utilization recorded [2.2, 3.8, 1.9, 11.5, 7.4, 1.9, 0.0, 0.0, 0.0, 0.0, 3.8, 5.6, 0.0, 1.9, 0.0, 1.9, 0.0]
2020-12-18 14:07:04,612 [update()] [INFO ]  Reading [57.0, 57.0, 58.0, 49.875, 49.875, 42.5, 44.85]
2020-12-18 14:07:04,612 [update()] [INFO ]  Reading [2284.919, 3426.598, 2364.726, 1986.223, 2018.7, 2092.788, 2495.559, 2089.821, 2258.047, 2923.834, 2140.166, 2003.691, 2003.955, 2057.83, 2300.01, 2082.556, 2314.2]
2020-12-18 14:07:04,612 [update()] [INFO ]  Reading [2.2, 3.8, 1.9, 11.5, 7.4, 1.9, 0.0, 0.0, 0.0, 0.0, 3.8, 5.6, 0.0, 1.9, 0.0, 1.9, 0.0]

```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. launch s-tui 
2. see "8-core processor" under CPU detector, despite all cores displayed in output.

What I see: 
[image]

amanusk, 2021-02-27:
I am not sure what the problem is.
In the screensot it seems that frequency and Uni all show 16 threads. What is missing?

cullyn-inverba, 2021-02-28:
Ah, I was misinformed when I made this. I assumed 16-core = 16 threads. My apologies.

closed 2021-02-28 by cullyn-inverba