amanusk/s-tui #169: Best font to get the amazing display like the screenshot

dpnova, 2021-01-31:
**Is your feature request related to a problem? Please describe.**
It seems hard to get s-tui to render as nice as the UI screenshot.

**Describe the solution you'd like**
Documentation of the font used to get such a nice render.

**Additional context**
Add any other context  about the feature request here. (screenshots, links to similar solutions)
[image]

amanusk, 2021-02-27:
I believe I used the default font used by [Alacritty](https://github.com/alacritty/alacritty) for the latest gif, and the Alacritty terminal.
Don't remember the color scheme unfortunately

clflush, 2021-05-16:
same 'issue'.

johnpiers, 2021-06-25:
Just so that I'm clear, you want it to look like the above screenshot? I use Tilix and when I run s-tui it looks like the above screenshot. My custom font that that I use "Fira Code Medium"
My colour scheme
[image]
[image]

closed 2025-05-13 by amanusk

dpnova, 2025-05-13:
@johnpiers the screenshot is of the horrible render i get using whatever it was at the time

currently I get this with Fira Code in kitty 

[image]

johnpiers, 2025-05-16:
Hi @dpnova 

To be honest I do not use s-tui anymore, it's been a while.  I've gravitated to a rather more complicated application called ATOP https://github.com/Atoptool/atop?tab=readme-ov-file that give a huge amount of information across just about any area of your PC and software.  Perhaps if you're not looking for anything complicated have a look some thing like BTOP++ https://github.com/aristocratos/btop  Cheers, hope that this is helpful.