amanusk/s-tui #171: Crash on raspberry pi after adding AMD power read

amanusk, 2021-02-27:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Rasbperry pi
  * OS version: _____
  * s-tui version: 1.1.1
  * Installation method(pip/PPA/source code): source
  
### Step 2: Describe the problem:
#### Observed Results:

Crashes when starting after adding and power read

```
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/s-tui/s_tui/s_tui.py", line 958, in <module>
    main()
  File "/home/pi/s-tui/s_tui/s_tui.py", line 912, in main
    graph_controller = GraphController(args)
  File "/home/pi/s-tui/s_tui/s_tui.py", line 716, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/home/pi/s-tui/s_tui/s_tui.py", line 651, in _load_config
    RaplPowerSource(),
  File "/home/pi/s-tui/s_tui/sources/rapl_power_source.py", line 45, in __init__
    self.reader = get_power_reader()
  File "/home/pi/s-tui/s_tui/sources/rapl_read.py", line 182, in get_power_reader
    if ReaderType.available():
  File "/home/pi/s-tui/s_tui/sources/rapl_read.py", line 165, in available
    if m[1] != "AuthenticAMD":
TypeError: 'NoneType' object is not subscriptable
```

### Step 3: Reproduce the problem:
#### Steps to reproduce:
Run `python3 -m s_tui.s_tui`

amanusk, 2021-02-27:
cc @schaerfo perhaps you know how to easily fix this?

JulianGro, 2021-09-13:
It this still up to date? Release v1.1.3 says `Fixed crashing bug on Raspberry pi`without any PR or issue linked.