amanusk/s-tui #172: Provide license file on PyPI

tim77, 2021-03-23:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): AMD
  * OS version: Fedora
  * s-tui version: 1.1.1
  * Installation method(pip/PPA/source code): DNF/RPM
  
### Step 2: Describe the problem:
#### Observed Results:

Currently `LICENSE` file available only on GitHub. Please provide also `LICENSE` file on PyPI. This is MUST [requirement](https://docs.fedoraproject.org/en-US/packaging-guidelines/LicensingGuidelines/#_license_text) for packaging into official repos in Fedora. Thanks!

amanusk, 2021-05-18:
I am not sure why I am not getting Email notifications on these issues. Sorry for the late response. I'll try to fix this asap

amanusk, 2021-07-24:
Hi @tim77 
Sorry again for the delay.
Is the current stat of PyPi satisfactory? https://pypi.org/project/s-tui/1.1.3/ 
If not, can you please show me an example of what additional information should be included?

tim77, 2021-07-24:
@amanusk hi. I did test build with 1.1.3 and License file available on PyPI. 👍 I think we can close this now.

Did you plan to tag new version on Github as well BTW?

amanusk, 2021-07-24:
Will do

amanusk, 2021-08-04:
@tim77 done