amanusk/s-tui #174: Crash on RISC-V

archanox, 2021-07-23:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Rvboards Nezha D1
  * OS version: Debian GNU/Linux 11 (bullseye)
  * s-tui version: 1.0.2
  * Installation method(pip/PPA/source code): apt
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
root@RVBoards:~# s-tui -d
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 33, in <module>
    sys.exit(load_entry_point('s-tui==1.0.2', 'console_scripts', 's-tui')())
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 903, in main
    graph_controller = GraphController(args)
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 715, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/usr/lib/python3/dist-packages/s_tui/s_tui.py", line 648, in _load_config
    FreqSource(),
  File "/usr/lib/python3/dist-packages/s_tui/sources/freq_source.py", line 50, in __init__
    self.top_freq = psutil.cpu_freq().max
AttributeError: 'NoneType' object has no attribute 'max'
root@RVBoards:~#  cat _s-tui.log
2021-07-23 14:32:48,555 [_load_config()] [DEBUG]  Config file not found
2021-07-23 14:32:48,556 [_load_config()] [DEBUG]  No refresh rate configed
2021-07-23 14:32:48,557 [_load_config()] [DEBUG]  No user config for utf8
2021-07-23 14:32:48,559 [_load_config()] [DEBUG]  No user config for temp threshold
2021-07-23 14:32:48,597 [__init__()] [DEBUG]  Temp sensor name Cpu_Thermal_Zone,0

```

amanusk, 2021-07-24:
Thanks for opening the issue and the interst in the project.

I would be very surprised it if would just work with RISC-V out of the box :), but the crash should not happen anyway.

Can you please try running with the latest version and report if the crash happens? Also, if by any chance you know how to impelement reading CPU frequency in RISC-V, PRs to here or to pstuil would be welcome.

archanox, 2023-05-26:
Tested successfully on the LicheePi4A

closed 2023-05-26 by archanox