amanusk/s-tui #176: Use Unicode Block Graphics U+2581-U+2588 for bar graphs

rbanffy, 2021-08-09:
**Is your feature request related to a problem? Please describe.**
The vertical resolution of the bar graphs is a character cell. It'd be 8x higher if using the block graphics between U+2581 and U+2588. Alternatively, horizontal and vertical resolutions can be doubled with 2x2 mosaics from U+2599 to U+259F. Optionally, depending on termina/font support, we could triple vertical resolution with the 2x3 mosaics introduced in [Unicode 13](https://en.wikipedia.org/wiki/Symbols_for_Legacy_Computing)

**Describe the solution you'd like**
Graphs would have a higher resolution

amanusk, 2021-08-09:
Sounds super useful. Could you be so kind to make a Pull Request for this?

rbanffy, 2021-08-09:
As soon as I figure out how the bars are drawn. Is this in this codebase or
is it something that comes from urwid?

amanusk, 2021-08-12:
Most of the parameters of urwid can be overridden, we do some of it in `sturwid`.
I guess this can be as well. @giltsuker is the expert on this honestly

giltsuker, 2021-08-12:
From what I can tell and remember a bit,
In bar_graph_vector.py, after we call the constructor of the native BarGraph class, I think you can override the native characters of the urwid base class.
override:
- self.eighths with the array for the top of the columns
- self.char with the "base" char for the rest of the column

Note that they should be an array I think for the 2 types of colors we use. Probably some trial and error will be needed :)

rbanffy, 2021-08-12:
Funny. You are not overriding the eights, but I saw the graph with full cells only... The eights **should** show.
[image]