amanusk/s-tui #179: NO power display

RuiYangQuan, 2021-12-24:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Raspberry Pi 
  * OS version: ubuntu18
  * s-tui version: 2.0
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:

  * What happened? (Traceback of a crash/Image/description):
  
I installed an S-TUi on the Raspberry PI, but no power display
```
* Replace this with debug output *
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. _____
2. _____
3. _____

```
* Replace this with relevant code to reproduce the problem *
```

amanusk, 2021-12-26:
Powe displays is only available on Intel and some Ryzen cpus.

sq-pyxisai, 2021-12-31:
I didn't see power until I ran s-tui as root.

amanusk, 2021-12-31:
Correct, power works with Intel and some Ryzen CPUs but tequires root to fetch the data.

KylinC, 2023-04-20:
sudo is all you need!

closed 2025-05-13 by amanusk