amanusk/s-tui #180: adjust frequency-graph max to reflect change in psutil reporting

sq-pyxisai, 2021-12-31:
**Is your feature request related to a problem? Please describe.**
Looks like `psutil v5.8.0` (used in `s-tui v1.0.0`) reported max frequency and instantaneous frequency in KHz, while `psutil v5.9.0` (latest version, and installed with current implementation of `s-tui`) reports max frequency in KHz but instantaneous frequency in MHz. Hence, the scale of  the frequency bar graph is such that frequency bars are too small to be seen.

**Describe the solution you'd like**
Modify function in `s_tui.sources.freq_source.py` as follows:
`    def get_top(self):`
        `logging.debug("Returning top %s", self.top_freq)`
        `# top frequency is in KHz, live frequency is in MHz, looks like. Need to divide top by 1000 so the`
        `# graph is scaled correctly`
        `return self.top_freq / 1000`

**Additional context**
May need some check of what version `psutil` is running and adjust this function accordingly...

sq-pyxisai, 2021-12-31:
my units might be wrong... it may be that 5.9 reports instantaneous frequency in GHz (and max frequency in MHz).

grobie, 2022-01-06:
Hah, missed this ticket here. I reported the issue in psutil: https://github.com/giampaolo/psutil/issues/2049 It appears that https://github.com/giampaolo/psutil/pull/1852 implemented caching and might have changed the behavior by that.

sq-pyxisai, 2022-01-07:
Nice. looks like the change made in `psutil` should fix it. I may try to pull that code later and see.

amanusk, 2022-01-07:
Thank you for openning the issue.
This seems to be a changed in the latest version of psutil. The fix is already implemented, but the package needs updating.
[#2049](https://github.com/giampaolo/psutil/commit/83d7067d7568f09ad95f094d17731e643a4a7ce6)

For now you can try manually installing the latest version of psutil

genema, 2022-03-07:
This issue, which is actually caused by 'psutil' == 5.9.0 can be solved by downgrade pstuil v5.8.0. However, I'm not very sure about the root of this problem currently.

ejgallego, 2022-08-24:
Installing psutil 5.9.1 solved the issue too.

sillyhatday, 2024-06-18:
I modified the source code to multiply by 1000. I came to report the issue and found this. The fix is indeed to update your version of psutil, not a problem with s-tui.

pip install --upgrade psutil

That was the fix