amanusk/s-tui #182: No Fequency Graphs on Ryzen 5950x

DocMAX, 2022-01-06:
s-tui 1.1.3
[image]

luoe, 2022-01-07:
Same behavior with weird low reported frequencies on an Intel Core i5-1135G7.
Other programs such as powertop or manually reading from 
```/sys/devices/system/cpu/cpufreq/policy{0..7}/scaling_cur_freq``` 
work as intended.

s-tui: 1.1.3
Linux: 5.15.13

amanusk, 2022-01-07:
Thanks for opening the issue. This seems to be the same problem as #180 
Frequency is reported in GHz, while Min/Max is reported in Mhz.

The change originates from a change in psutil, the fix for which should be already in the works.

I suggest we wait for the psutil

luoe, 2022-01-12:
@DocMAX if you use Archlinux/Manjaro, you can try do downgrade python-psutil to 5.8.0. This resolves this issue for now and only one core showing the exact frequency:
```
sudo pacman -U https://archive.archlinux.org/packages/p/python-psutil/python-psutil-5.8.0-4-x86_64.pkg.tar.zst
```

And add 
```
IgnorePkg   = python-psutil
```
to pacman.conf under [options].

genema, 2022-03-07:
This can be solved by downgrade psutil to v5.8.0 .

brunodrugowick, 2022-08-13:
In Ubuntu-based distros, this should fix, but I don't know if it has side effects:

```
pip install --upgrade psutil==5.8.0
```

ejgallego, 2022-08-24:
Duplicate of #180