amanusk/s-tui #186: S-tui shows unit as Mhz, even tho the value is in GHz

neilschark, 2022-02-02:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): AMD R9 3900X
  * OS version: Arch Linux
  * s-tui version: 1.1.3 / master
  * Installation method(pip/PPA/source code): pacman, source code
  
### Step 2: Describe the problem:
#### Observed Results:

s-tui shows the CPU frequency as GHz, although is shows the data in Ghz.

[image]

  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
* Replace this with debug output *
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. open s-tui

```
* Replace this with relevant code to reproduce the problem *
```

1202software, 2022-02-06:
Also, besides the summary showing 'MHz' instead of 'GHz', the graphs for frequency are empty as they are likely interpreting it as MHz instead of GHz.

neilschark, 2022-02-06:
Than it seems like the intended unit is MHz indeed and it's a problem of the reading of the data.

ravindk89, 2022-02-09:
Also having this issue on s-tui 1.1.3, same rough setup (Manjaro 21.2.2 on a Thinkpad X1 Carbon gen 9)

I also noticed `glances` isn't showing CPU frequency either. `lscpu` also isn't showing current MHz. I can see it via `cat /proc/cpuinfo`. 

I'm wondering if something in arch/manjaro changed where reading CPU frequency no longer works as expected

@neilschark any chance you're using `cpupower` to set the CPU states + ranges?

hook77, 2022-02-15:
Hello, 
I have the same issue on my Ryzen 5 PRO on Ubuntu Linux with kernel 5.13.
I did some quick code inspection, and I removed the 1000 divider in the psutils package in the _pslinux.py file. 
But looks like this is a bug in the psutil library, and not an s-tui issue, or?

hot22shot, 2022-03-11:
Hi, 

Same issue with s-tui version 1.1.3 on unraid running Kernel 5.15.27.
My CPU is a Ryzen 5800X.

ohagandavid, 2022-03-13:
I just installed s-tui today on my i5-4690K. Did not have the frequency graphs. Made the change hook77 made to _pslinux.py and the psutils.cpu_freq() function and s-tui now display in Mhz and everything looks golden.

Oisangui, 2022-03-20:
@hook77 @ohagandavid Could you be more detailed about your solution? Where is that file located?

ohagandavid, 2022-03-21:
Was ~/.local/lib/python3.9/site-packages/psutil/_pslinux.py for me. At line 762 in the cpu_freq() function I've commented out `curr = int(curr) / 1000`.

michaellass, 2022-04-16:
This should get fixed when psutil 5.9.1 is released:
https://github.com/giampaolo/psutil/commit/83d7067d7568f09ad95f094d17731e643a4a7ce6

wc7086, 2022-04-20:
```bash
psutilVersion=`pip show psutil | grep 'Version' | awk -F ': ' '{print $2}'` \
&& psutilLocation=`pip show psutil | grep 'Location' | awk -F ': ' '{print $2}'`/psutil \
&& [ $psutilVersion == 5.9.0 ] \
&& [[ `sed -n 752p $psutilLocation/_pslinux.py | grep '1000'` == '' ]] \
&& sudo sed -i '752 s/$/ * 1000/' $psutilLocation/_pslinux.py
```

SuperKenVery, 2022-07-10:
This problem has been solved in psutil v5.9.1 (5.9.0 still has this problem)

So I suggest that we set the required psutil to be `>=5.9.1`?

amanusk, 2022-07-10:
Yep, definitely we should.
My worry is mainly the way debian/ubuntu is packaged, and that this will not go away until psutil is updated there, but should be updated regardless.

EHJ-52n, 2022-07-18:
The solution from [wc7086](#issuecomment-1104379530) worked for me, but some limitations as I am using zsh:

```shell
~ psutilVersion=`pip show psutil | grep 'Version' | awk -F ': ' '{print $2}'` && echo $psutilVersion
5.9.0
~ pip show psutil | grep 'Location' | awk -F ': ' '{print $2}'        
/usr/lib/python3/dist-packages
~ psutilVersion=`pip show psutil | grep 'Version' | awk -F ': ' '{print $2}'` \
&& psutilLocation=`pip show psutil | grep 'Location' | awk -F ': ' '{print $2}'`/psutil \
&& [ $psutilVersion == 5.9.0 ] \
&& sudo sed -i "/curr = cpuinfo_freqs\[i\]$/ s/$/ * 1000/" $psutilLocation/_pslinux.py
zsh: = not found
~ sudo sed -i "/curr = cpuinfo_freqs\[i\]$/ s/$/ * 1000/" $psutilLocation/_pslinux.py
```

After applying this patch, the frequency graph is rendered as usual.

My set-up is the following:

```shell
~ lsb_release -d | awk -F ':' '{ print $2}' \
&& python --version \
&& echo "psutil: $(pip show psutil | grep "Version" | awk '{print $2}')" \
&& s-tui --version
	Ubuntu 22.04 LTS
Python 3.10.4
psutil: 5.9.0
s-tui 1.1.3 - (C) 2017-2020 Alex Manuskin, Gil Tsuker
    Released under GNU GPLv2
```

commit c61db5ee (2022-07-30): Bump psutil dependency to min 5.9.1 to fix #186
closed 2022-07-30 by amanusk