amanusk/s-tui #189: pip3 install s-tui failed (error: metadata-generation-failed)

FoundAdd, 2022-03-08:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel
  * OS version: CentOS 7
  * s-tui version: 1.1.3
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:
`install s-tui failed`<br/>
`I previously installed s-tui 1.1.3 in a Python 3.6 environment and later installed Python 3.7, so I uninstalled Python 3.6. Later, when running s-tui, I was told that I could not find the file, so I searched all the files containing "s-tui" under "/", but only found "/usr/local/bin/s-tui". I judged that s-tui was deleted together in the lib folder of python 3.6 when I uninstalled python 3.6, so I plan to reinstall s-tui. Finally this error occurred when reinstalling s-tui.`
  * What happened? (Traceback of a crash/Image/description):
  ```
  pip3 install s-tui
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Looking in indexes: https://pypi.python.org/simple
Collecting s-tui
  Using cached https://files.pythonhosted.org/packages/0c/94/23d2062ab57fe78fa886e162c14a70f4c2bbd94961aa762074ed8ca36fb6/s-tui-1.1.3.tar.gz (41 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      ERROR: Can not execute `setup.py` since setuptools is not available in the build environment.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
  ```
  
#### Debug Results, output of `s-tui -d` created in a file `_s-tui.log`: 
```
no log
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Input "pip3 install s-tui" in shell.
2. error.

Manamama, 2024-05-31:
Just try:
```
pip3 install setuptools
pip3 install --upgrade pip
pip3 install s-tui
```

FoundAdd, 2024-06-01:
> Just try:
> 
> ```
> pip3 install setuptools
> pip3 install --upgrade pip
> pip3 install -U s-tui
> ```

Thank you for your response. It seems effective, but I had to reinstall my server last year due to a disk failure, so I can't verify it. Fortunately, this issue no longer exists after the system reinstallation.

closed 2024-06-01 by FoundAdd