amanusk/s-tui #195: No wattage displayed for package

davidhealey, 2022-07-17:
The graph is populated but the value is always 0

[image]

amanusk, 2022-07-18:
Could you please show the full screenshots with the values at 0?

davidhealey, 2022-07-18:
[image]

davidhealey, 2022-07-18:
I want to see how much power (watts) the cpu is consuming.

amanusk, 2022-07-18:
Oh, The menu on the left is scrollable, just scroll it down to the power section,
with down-arrow, page down or j key

[image]

davidhealey, 2022-07-18:
Ah ok yes I'm seeing that now, thank you. It looks very low, it's showing 1.9W for the package, is that all cores together?

amanusk, 2022-07-18:
It's not unreasonable, depending on the processor and on the load.
At idle this is a reasonable number for the entire processor.
The numbers are probed from Intel RAPL, so I hope it is reported correctly :)