amanusk/s-tui #197: The idle frequency show by cpupower and s-tui not same.

edisonchan, 2022-08-02:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel 11700K/12900KF, c-state disabled in BIOS, and the cpu ratio was set to 40.0x
  * OS version: Ubuntu 22.04 kernel 5.19
  * s-tui version: latest from git clone (c61db5e)
  * Installation method(pip/PPA/source code): source core
  
### Step 2: Describe the problem:
#### Observed Results:
cpupower 5.19:
[image]

s-tui:
[image]

  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. set CPU ratio to 40.x, sync All core, disable C-state in BIOS
2. run cpupower: 
watch -n 1 sudo cpupower monitor
4. run stui:
s-tui

ronny-rentner, 2022-10-19:
I have a similar issue with s-tui vs i7z showing different frequencies.

AlexeiScherbakov, 2022-12-01:
On Rock PI 4 s-tui display correct frequencies, but on Raspberry PI 4 600 MHz shows as 1500 MHz:
```
Frequency      [MHz]|                                   Util [%]
 Avg           1500.0|100       Avg     |    Core 0    |    Core 1    |    Core 2    |    Core 3
 Core 0        1500.0|                  |              |              |              |
 Core 1        1500.0|                  |              |              |              |
 Core 2        1500.0|                  |              |              |              |
 Core 3        1500.0|                  |              |              |              |
                     |                  |              |              |              |
 Util             [%]|                  |              |              |              |
 Avg              3.2|                  |              |              |              |
 Core 0           0.0|                  |              |              |              |
pi@pi:/sys/devices/system/cpu $ sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq
600000
pi@pi:/sys/devices/system/cpu $ sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq
600000
pi@pi:/sys/devices/system/cpu $ sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq
1500000
pi@pi:/sys/devices/system/cpu $
```