amanusk/s-tui #201: s-tui crashes if some cores are offline

rkraneis, 2022-12-05:
### Step 1: Describe your environment

  * System (Intel/AMD/Raspberry Pi etc): Intel i7-12700H
  * OS version: Fedora 37
  * s-tui version: 1.1.4
  * Installation method(pip/PPA/source code): dnf
  
### Step 2: Describe the problem:
#### Observed Results:

When taking some of the cores offline, s-tui just crashes:
```
s-tui -d
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 33, in <module>
    sys.exit(load_entry_point('s-tui==1.1.4', 'console_scripts', 's-tui')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 912, in main
    graph_controller.main()
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 757, in main
    loop.run()
  File "/usr/lib64/python3.11/site-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib64/python3.11/site-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib64/python3.11/site-packages/urwid/main_loop.py", line 789, in run
    self._loop()
  File "/usr/lib64/python3.11/site-packages/urwid/main_loop.py", line 822, in _loop
    alarm_callback()
  File "/usr/lib64/python3.11/site-packages/urwid/main_loop.py", line 172, in cb
    callback(self, user_data)
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 853, in animate_graph
    self.view.update_displayed_information()
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 249, in update_displayed_information
    source.update()
  File "/usr/lib/python3.11/site-packages/s_tui/sources/freq_source.py", line 62, in update
    self.last_measurement = [psutil.cpu_freq(False).current]
                             ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/site-packages/psutil/__init__.py", line 1864, in cpu_freq
    ret = _psplatform.cpu_freq()
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/site-packages/psutil/_pslinux.py", line 742, in cpu_freq
    raise NotImplementedError(
```

It also doesn't start until all cores are online again:
```
s-tui 
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 33, in <module>
    sys.exit(load_entry_point('s-tui==1.1.4', 'console_scripts', 's-tui')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 911, in main
    graph_controller = GraphController(args)
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 715, in __init__
    possible_sources = self._load_config(args.t_thresh)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 648, in _load_config
    FreqSource(),
    ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/s_tui/sources/freq_source.py", line 45, in __init__
    self.last_measurement = [0] * len(psutil.cpu_freq(True))
                                      ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/site-packages/psutil/__init__.py", line 1864, in cpu_freq
    ret = _psplatform.cpu_freq()
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/site-packages/psutil/_pslinux.py", line 742, in cpu_freq
    raise NotImplementedError(
```
[... 76 lines cut ...]
2022-12-05 08:54:43,174 [update()] [INFO ]  Core id util 25.6
2022-12-05 08:54:43,174 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,174 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,174 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,174 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,174 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,175 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,175 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,175 [update()] [INFO ]  Core id util 100.0
2022-12-05 08:54:43,175 [update()] [INFO ]  Utilization recorded [75.4, 100.0, 24.4, 100.0, 7.0, 100.0, 14.4, 100.0, 14.9, 100.0, 18.3, 99.5, 25.6, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0]
2022-12-05 08:54:43,175 [update()] [INFO ]  Reading [78.0, 78.0, 78.0, 78.0, 78.0, 77.0, 77.0, 77.0, 77.0, 74.0, 75.0, 74.0, 77.0, 74.0, 77.0, 78.0, 78.0, 30.85, 30.85, 35.85]
2022-12-05 08:54:43,176 [update()] [INFO ]  Reading [2951.3931999999995, 3100.031, 3100.0, 3100.0, 3182.317, 3100.031, 3100.0, 3100.0, 3100.0, 3100.0, 3100.0, 3100.0, 3100.0, 2700.052, 2700.007, 2700.09, 2700.016, 2700.046, 2699.987, 2699.916, 2700.066]
2022-12-05 08:54:43,176 [update()] [INFO ]  Reading [75.4, 100.0, 24.4, 100.0, 7.0, 100.0, 14.4, 100.0, 14.9, 100.0, 18.3, 99.5, 25.6, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0]
2022-12-05 08:54:45,199 [eval_hooks()] [DEBUG]  Evaluating hooks
```
  
### Step 3: Reproduce the problem:
#### Steps to reproduce:
1. Run s-tui
2. disable some cores
3. observe crash

```
echo 0 | sudo tee /sys/devices/system/cpu/cpu0/online
```

FallingSnow, 2023-02-01:
I'm experiencing the same issue. Disable some cores than try to run s-tui.
```
archlinux% sudo s-tui
Traceback (most recent call last):
  File "/usr/bin/s-tui", line 33, in <module>
    sys.exit(load_entry_point('s-tui==1.1.4', 'console_scripts', 's-tui')())
  File "/usr/lib/python3.10/site-packages/s_tui/s_tui.py", line 912, in main
    graph_controller = GraphController(args)
  File "/usr/lib/python3.10/site-packages/s_tui/s_tui.py", line 716, in __init__
    possible_sources = self._load_config(args.t_thresh)
  File "/usr/lib/python3.10/site-packages/s_tui/s_tui.py", line 649, in _load_config
    FreqSource(),
  File "/usr/lib/python3.10/site-packages/s_tui/sources/freq_source.py", line 46, in __init__
    self.last_measurement = [0] * len(psutil.cpu_freq(True))
  File "/usr/lib/python3.10/site-packages/psutil/__init__.py", line 1864, in cpu_freq
    ret = _psplatform.cpu_freq()
  File "/usr/lib/python3.10/site-packages/psutil/_pslinux.py", line 742, in cpu_freq
    raise NotImplementedError(
NotImplementedError: can't find current frequency file
```

ohsix, 2023-05-14:
same, don't know if this can be fixed in s-tui though, probably needs to be fixed in psutil

https://github.com/giampaolo/psutil/blob/7eadee31db2f038763a3a6f978db1ea76bbc4674/psutil/_pslinux.py#L749 needs an extra check to curr to skip cpus without frequency files

Jamesits, 2024-01-23:
Link the relevant issue here: https://github.com/giampaolo/psutil/issues/2254

Betaminos, 2024-07-29:
> Link the relevant issue here: [giampaolo/psutil#2254](https://github.com/giampaolo/psutil/issues/2254)

It seems the upstream issue has been fixed by now, thanks for linking to it 👍

amanusk, 2025-05-13:
Should be closed in newer psutil versions. Closing for now

closed 2025-05-13 by amanusk