amanusk/s-tui #212: Incompatibility with `urwid>=2.4.3`

randolf-scholz, 2024-01-21:
### Step 1: Describe your environment

  * OS version: Ubuntu 22.04
  * s-tui version: 1.1.4
  * Installation method(pip/PPA/source code): pip
  
### Step 2: Describe the problem:
#### Observed Results:

Install simply via `pip install s-tui` (should install `urwid` 2.4.4)

<details> <summary> Trying to open the graphs tab produces an error </summary>

```python
Traceback (most recent call last):
  File "/home/rscholz/.pyenv/versions/default/bin/s-tui", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/s_tui/s_tui.py", line 913, in main
    graph_controller.main()
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/s_tui/s_tui.py", line 758, in main
    loop.run()
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/event_loop/main_loop.py", line 343, in run
    self._run()
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/event_loop/main_loop.py", line 445, in _run
    self.event_loop.run()
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/event_loop/select_loop.py", line 182, in run
    self._loop()
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/event_loop/select_loop.py", line 229, in _loop
    record.data()
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/display/_posix_raw_display.py", line 273, in wrapper
    return self.parse_input(event_loop, callback, self.get_available_raw_input())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/display/_raw_display_base.py", line 493, in parse_input
    callback(decoded_codes, raw_codes)
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/event_loop/main_loop.py", line 472, in _update
    self.process_input(keys)
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/event_loop/main_loop.py", line 573, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)  # noqa: PLW2901
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/widget.py", line 798, in keypress
    return get_delegate(self).keypress(size, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/padding.py", line 418, in keypress
    return self._original_widget.keypress(maxvals, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/columns.py", line 1089, in keypress
    key = w.keypress(size_args[i], key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/s_tui/sturwid/ui_elements.py", line 92, in keypress
    return super(ViListBox, self).keypress(size, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/listbox.py", line 1103, in keypress
    key = focus_widget.keypress((maxcol,), key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/grid_flow.py", line 492, in keypress
    key = super().keypress(size, key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/widget.py", line 798, in keypress
    return get_delegate(self).keypress(size, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/pile.py", line 841, in keypress
    key = self.focus.keypress(size_args[i], key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/padding.py", line 418, in keypress
    return self._original_widget.keypress(maxvals, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/columns.py", line 1089, in keypress
    key = w.keypress(size_args[i], key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/widget.py", line 798, in keypress
    return get_delegate(self).keypress(size, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/wimp.py", line 762, in keypress
    self._emit("click")
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/widget.py", line 428, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/signals.py", line 301, in emit
    result |= self._call_callback(callback, user_arg, weak_args, user_args, args)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/signals.py", line 325, in _call_callback
    return bool(callback(*args))
                ^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/s_tui/s_tui.py", line 354, in on_graphs_menu_open
    self.original_widget = urwid.Overlay(
                           ^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/overlay.py", line 166, in __init__
    self.set_overlay_parameters(align, width, valign, height, min_width, min_height, left, right, top, bottom)
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/overlay.py", line 455, in set_overlay_parameters
    self.options(
  File "/home/rscholz/.pyenv/versions/3.11.6/envs/default/lib/python3.11/site-packages/urwid/widget/overlay.py", line 355, in options
    Align(align_type),
    ^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/lib/python3.11/enum.py", line 712, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rscholz/.pyenv/versions/3.11.6/lib/python3.11/enum.py", line 1135, in __new__
    raise ve_exc
ValueError: <WHSettings.RELATIVE: 'relative'> is not a valid Align
```

</details>

The error does not occur if `urwid<=2.4.2` is installed.

amanusk, 2024-01-26:
Thank you for the report.
Please check with the latest version

randolf-scholz, 2024-01-28:
👍 Works.

closed 2024-01-28 by randolf-scholz