amanusk/s-tui #219: AttributeError: module ‘zmq’ has no attribute ‘POLLIN’ when running s-tui, on Termux

Manamama, 2024-05-31:
Bug Report: s-tui compatibility issue with pyzmq on Termux (Android)

Description:
When attempting to run the newest s-tui system monitoring tool on Termux (an Android terminal emulator), even after full pip force reinstall, the application fails to start with the following error:

```
Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/bin/s-tui", line 5, in <module>
    from s_tui.s_tui import main
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/s_tui/s_tui.py", line 38, in <module>
    import urwid
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/urwid/__init__.py", line 97, in <module>
    from . import display, event_loop, widget
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/urwid/event_loop/__init__.py", line 51, in <module>
    from .zmq_loop import ZMQEventLoop
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/urwid/event_loop/zmq_loop.py", line 52, in <module>
    class ZMQEventLoop(EventLoop):
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/urwid/event_loop/zmq_loop.py", line 130, in ZMQEventLoop
    flags: int = zmq.POLLIN,
                 ^^^^^^^^^^
AttributeError: module 'zmq' has no attribute 'POLLIN'
```

This error occurs due to a compatibility issue between the urwid library (used by s-tui) and the version of the pyzmq package (Python bindings for the ZeroMQ library) installed on the Termux environment.

Environment:
- Operating System: Android (Termux terminal emulator)
- Linux kernel version: 4.14.186+
- Python version: 3.11
- s-tui version: (latest from PyPI)
- urwid version: (latest from PyPI)
- pyzmq version: (initially 25.1.2, later updated to 26.0.3)

Steps to Reproduce:
1. Install s-tui on Termux using `pip install s-tui`
2. Run `s-tui` from the terminal

Expected Behavior:
The s-tui application should start and display the system monitoring interface.

Actual Behavior:
The application fails to start with the `AttributeError: module 'zmq' has no attribute 'POLLIN'` error.

Solution:
The issue was resolved by forcing a reinstallation of the pyzmq package with the following command:

```
pip install --pre --force-reinstall pyzmq
```

This command uninstalled the existing version of pyzmq (25.1.2) and installed the pre-release version 26.0.3, which resolved the compatibility issue with the urwid library.

After the forced reinstallation of pyzmq, running `s-tui` successfully started the application without any errors.

Additional Notes:
- The root cause of the issue appears to be a mismatch between the version of the urwid library and the version of the pyzmq package installed on the Termux environment.
- The forced reinstallation process likely resolved any conflicts or issues with the previous installation of pyzmq.
- It is recommended to investigate and address the underlying compatibility issue between urwid and pyzmq to ensure a more robust and reliable solution, maybe adding these to requirements, so that force reinstall of s-tui fixes it.

Manamama, 2024-06-10:
Solved: I have had incompatible pyzmq packages: careful removal of all related libraries and reinstalling just one helped.

closed 2024-06-10 by Manamama

amanusk, 2024-06-10:
@Manamama thank you for following up and updating on this issue :pray: